LudovicRousseau/CCID #144: Test communicate with SAM SEC1210 module via UART on ANdroid?

tuananh-lab, 2024-10-14:
My board contain module SAM SEC1210 only support UART.
My board run Android 13 OS
I created /dev/ttyHS5 via 2 GPIO ports via schematic to communicate with module SAM.
Is there any command or apk I can install to communicate with the SAM module and get information from the SAM card or sim card?
Now what can I do next?

LudovicRousseau, 2024-10-14:
No idea.
I do not use Android as a target.

I suggest to ask on pcsclite-muscle mailing list https://lists.infradead.org/mailman/listinfo/pcsclite-muscle

tuananh-lab, 2024-10-22:
@LudovicRousseau I want to test module SAM SEC1210 with sim card via UART using pcscd and pcsc_scan but no detect sim card. I try send APDU command PC_to_RDR_IccPowerOff
	TX:03 06 63 00 00 00 00 00 01 00 00 00 67
	RX:03 06 81 00 00 00 00 00 01 02 00 01 86
It look error because don't see ATR.
Module SAM SEC1210 do work with SIM card ?
 Are you help me?