LudovicRousseau/CCID #145: Broadcom 58200 0A5C:5864 Card Not Present

hgwheels, 2024-10-17:
Broadcom 58200 0A5C:5864 never detects a card as present in the reader, despite being physically inserted.  I have repeated this behavior on multiple physical readers of the same model in different physical Latitude 5450.

Ubuntu 22.04.4
Dell Latitude 5450
Broadcom 58200 05AC:5864

[log.txt](https://github.com/user-attachments/files/17411927/log.txt)
[output.bin.zip](https://github.com/user-attachments/files/17412017/output.bin.zip)

opensc-tool --list-readers 
P:5028; T:0x138491002771456 07:52:00.572 [opensc-tool] ctx.c:855:sc_context_create: ===================================
P:5028; T:0x138491002771456 07:52:00.572 [opensc-tool] ctx.c:856:sc_context_create: opensc version: 0.22.0
P:5028; T:0x138491002771456 07:52:00.572 [opensc-tool] reader-pcsc.c:886:pcsc_init: PC/SC options: connect_exclusive=0 disconnect_action=0 transaction_end_action=0 reconnect_action=0 enable_pinpad=1 enable_pace=1
P:5028; T:0x138491002771456 07:52:00.573 [opensc-tool] reader-pcsc.c:1386:pcsc_detect_readers: called
P:5028; T:0x138491002771456 07:52:00.573 [opensc-tool] reader-pcsc.c:1399:pcsc_detect_readers: Probing PC/SC readers
P:5028; T:0x138491002771456 07:52:00.573 [opensc-tool] reader-pcsc.c:1452:pcsc_detect_readers: Establish PC/SC context
P:5028; T:0x138491002771456 07:52:00.745 [opensc-tool] reader-pcsc.c:1335:pcsc_add_reader: Adding new PC/SC reader 'Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00'
P:5028; T:0x138491002771456 07:52:00.745 [opensc-tool] reader-pcsc.c:360:refresh_attributes: Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00 check
P:5028; T:0x138491002771456 07:52:00.745 [opensc-tool] reader-pcsc.c:407:refresh_attributes: current  state: 0x00000012
P:5028; T:0x138491002771456 07:52:00.745 [opensc-tool] reader-pcsc.c:408:refresh_attributes: previous state: 0x00000000
P:5028; T:0x138491002771456 07:52:00.745 [opensc-tool] reader-pcsc.c:462:refresh_attributes: card absent
P:5028; T:0x138491002771456 07:52:00.750 [opensc-tool] reader-pcsc.c:1544:pcsc_detect_readers: Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00:SCardConnect(DIRECT): 0x00000000
P:5028; T:0x138491002771456 07:52:00.750 [opensc-tool] reader-pcsc.c:1152:detect_reader_features: called
P:5028; T:0x138491002771456 07:52:00.750 [opensc-tool] reader-pcsc.c:1154:detect_reader_features: Requesting reader features ... 
P:5028; T:0x138491002771456 07:52:00.750 [opensc-tool] reader-pcsc.c:1175:detect_reader_features: Reader feature 12 found
P:5028; T:0x138491002771456 07:52:00.750 [opensc-tool] reader-pcsc.c:1082:part10_detect_max_data: get dwMaxAPDUDataSize property returned 65536
P:5028; T:0x138491002771456 07:52:00.750 [opensc-tool] reader-pcsc.c:1285:detect_reader_features: Reader supports transceiving 65536 bytes of data
P:5028; T:0x138491002771456 07:52:00.750 [opensc-tool] reader-pcsc.c:1290:detect_reader_features: Sending is limited to 255 bytes of data in configuration file
P:5028; T:0x138491002771456 07:52:00.750 [opensc-tool] reader-pcsc.c:1295:detect_reader_features: Receiving is limited to 256 bytes of data in configuration file
P:5028; T:0x138491002771456 07:52:00.751 [opensc-tool] reader-pcsc.c:1121:part10_get_vendor_product: id_vendor=0a5c id_product=5864
P:5028; T:0x138491002771456 07:52:00.751 [opensc-tool] reader-pcsc.c:1568:pcsc_detect_readers: returning with: 0 (Success)
# Detected readers (pcsc)
Nr.  Card  Features  Name
P:5028; T:0x138491002771456 07:52:00.751 [opensc-tool] sc.c:335:sc_detect_card_presence: called
P:5028; T:0x138491002771456 07:52:00.751 [opensc-tool] reader-pcsc.c:472:pcsc_detect_card_presence: called
P:5028; T:0x138491002771456 07:52:00.751 [opensc-tool] reader-pcsc.c:360:refresh_attributes: Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00 check
P:5028; T:0x138491002771456 07:52:00.752 [opensc-tool] reader-pcsc.c:385:refresh_attributes: returning with: 0 (Success)
P:5028; T:0x138491002771456 07:52:00.752 [opensc-tool] reader-pcsc.c:477:pcsc_detect_card_presence: returning with: 0 (Success)
P:5028; T:0x138491002771456 07:52:00.752 [opensc-tool] sc.c:340:sc_detect_card_presence: returning with: 0 (Success)
0    No              Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00
P:5028; T:0x138491002771456 07:52:00.752 [opensc-tool] ctx.c:931:sc_release_context: called
P:5028; T:0x138491002771456 07:52:00.752 [opensc-tool] reader-pcsc.c:974:pcsc_finish: called

opensc-tool --reader 0 --atr
P:5047; T:0x133099394861056 07:52:31.467 [opensc-tool] ctx.c:855:sc_context_create: ===================================
P:5047; T:0x133099394861056 07:52:31.467 [opensc-tool] ctx.c:856:sc_context_create: opensc version: 0.22.0
P:5047; T:0x133099394861056 07:52:31.467 [opensc-tool] reader-pcsc.c:886:pcsc_init: PC/SC options: connect_exclusive=0 disconnect_action=0 transaction_end_action=0 reconnect_action=0 enable_pinpad=1 enable_pace=1
P:5047; T:0x133099394861056 07:52:31.467 [opensc-tool] reader-pcsc.c:1386:pcsc_detect_readers: called
P:5047; T:0x133099394861056 07:52:31.467 [opensc-tool] reader-pcsc.c:1399:pcsc_detect_readers: Probing PC/SC readers
P:5047; T:0x133099394861056 07:52:31.467 [opensc-tool] reader-pcsc.c:1452:pcsc_detect_readers: Establish PC/SC context
P:5047; T:0x133099394861056 07:52:31.475 [opensc-tool] reader-pcsc.c:1335:pcsc_add_reader: Adding new PC/SC reader 'Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00'
P:5047; T:0x133099394861056 07:52:31.475 [opensc-tool] reader-pcsc.c:360:refresh_attributes: Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00 check
P:5047; T:0x133099394861056 07:52:31.476 [opensc-tool] reader-pcsc.c:407:refresh_attributes: current  state: 0x00000012
P:5047; T:0x133099394861056 07:52:31.476 [opensc-tool] reader-pcsc.c:408:refresh_attributes: previous state: 0x00000000
P:5047; T:0x133099394861056 07:52:31.476 [opensc-tool] reader-pcsc.c:462:refresh_attributes: card absent
P:5047; T:0x133099394861056 07:52:31.482 [opensc-tool] reader-pcsc.c:1544:pcsc_detect_readers: Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00:SCardConnect(DIRECT): 0x00000000
P:5047; T:0x133099394861056 07:52:31.482 [opensc-tool] reader-pcsc.c:1152:detect_reader_features: called
P:5047; T:0x133099394861056 07:52:31.482 [opensc-tool] reader-pcsc.c:1154:detect_reader_features: Requesting reader features ... 
P:5047; T:0x133099394861056 07:52:31.482 [opensc-tool] reader-pcsc.c:1175:detect_reader_features: Reader feature 12 found
P:5047; T:0x133099394861056 07:52:31.482 [opensc-tool] reader-pcsc.c:1082:part10_detect_max_data: get dwMaxAPDUDataSize property returned 65536
P:5047; T:0x133099394861056 07:52:31.482 [opensc-tool] reader-pcsc.c:1285:detect_reader_features: Reader supports transceiving 65536 bytes of data
P:5047; T:0x133099394861056 07:52:31.482 [opensc-tool] reader-pcsc.c:1290:detect_reader_features: Sending is limited to 255 bytes of data in configuration file
P:5047; T:0x133099394861056 07:52:31.482 [opensc-tool] reader-pcsc.c:1295:detect_reader_features: Receiving is limited to 256 bytes of data in configuration file
P:5047; T:0x133099394861056 07:52:31.482 [opensc-tool] reader-pcsc.c:1121:part10_get_vendor_product: id_vendor=0a5c id_product=5864
P:5047; T:0x133099394861056 07:52:31.483 [opensc-tool] reader-pcsc.c:1568:pcsc_detect_readers: returning with: 0 (Success)
P:5047; T:0x133099394861056 07:52:31.485 [opensc-tool] sc.c:335:sc_detect_card_presence: called
P:5047; T:0x133099394861056 07:52:31.485 [opensc-tool] reader-pcsc.c:472:pcsc_detect_card_presence: called
P:5047; T:0x133099394861056 07:52:31.485 [opensc-tool] reader-pcsc.c:360:refresh_attributes: Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00 check
P:5047; T:0x133099394861056 07:52:31.487 [opensc-tool] reader-pcsc.c:385:refresh_attributes: returning with: 0 (Success)
P:5047; T:0x133099394861056 07:52:31.487 [opensc-tool] reader-pcsc.c:477:pcsc_detect_card_presence: returning with: 0 (Success)
[... 87 lines cut ...]
   No special characteristics
  dwFeatures: 0x000102BA
   ....02 Automatic parameter configuration based on ATR data
   ....08 Automatic ICC voltage selection
   ....10 Automatic ICC clock frequency change according to parameters
   ....20 Automatic baud rate change according to frequency and Fi, Di params
   ....80 Automatic PPS made by the CCID
   ..02.. NAD value other than 00 accepted (T=1)
   01.... TPDU level exchange
  dwMaxCCIDMessageLength: 271 bytes
  bClassGetResponse: 0x00
  bClassEnvelope: 0x00
  wLcdLayout: 0x0000
  bPINSupport: 0x00
  bMaxCCIDBusySlots: 1

pcsc_scan -v -t 30
Using reader plug'n play mechanism
Scanning present readers...
0: Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00
 
Thu Oct 17 07:54:56 2024
 Reader 0: Broadcom Corp 58200 [Contacted SmartCard] (0123456789ABCD) 00 00
  Event number: 0
  Card state: Card removed,

LudovicRousseau, 2024-10-17:
In my list this reader has 2 interfaces: contact and contactless. https://ccid.apdu.fr/ccid/readers/Broadcom_58200_0x5864.txt
Your reader only has the contact interface.

Check if you can find a UEFI/BIOS update from Dell.

I guess the reader+card works fine on Windows. Exact?

hgwheels, 2024-10-17:
As far as I can tell this reader only has one interface (there is no marking on the laptop housing indicating contactless capability)

The laptop and logs above are running the latest Dell UEFI version 1.6.0

Correct the reader and card combo work exactly as expected on Windows

https://www.dell.com/support/home/en-tc/drivers/driversdetails?driverid=1pw34&oscode=biosa&productcode=latitude-14-5450-laptop

sudo dmidecode -s bios-version
1.6.0

sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 3 local devices supported

sudo fwupdmgr get-updates
Devices with no available firmware updates: 
 • UEFI dbx
Devices with the latest available firmware version:
 • BG6 KIOXIA 256GB
 • System Firmware
 • UEFI Device Firmware
No updates available

sudo fwupdmgr update
Devices with no available firmware updates: 
 • UEFI dbx
Devices with the latest available firmware version:
 • BG6 KIOXIA 256GB
 • System Firmware
 • UEFI Device Firmware

hgwheels, 2024-10-17:
From the Dell build sheet:

Part Number: 185D2
Description: INFO,SC,CONTACT

LudovicRousseau, 2024-10-17:
I have no easy solution.
I will need to get a Dell Latitude 5450 and do some more analysis.

LudovicRousseau, 2024-11-12:
@hgwheels can you provide me a Dell Latitude 5450 for my tests?
If not I can't do much on this issue.

LudovicRousseau, 2025-02-01:
I do not have the hardware to reproduce and work on the issue.
I suggest you to contact the laptop vendor: Dell.

closed 2025-02-01 by LudovicRousseau