LudovicRousseau/CCID #146: Update to CCID requires change in GPG configuration?

colemickens, 2024-10-20:
Hi,

Apologies up front, I'm not sure if this is actually a bug in CCID, but I'm a bit desperate for some pointers.

Some NixOS users noticed that their Yubikey/CCID/gnupg setups have broken (again): https://github.com/NixOS/nixpkgs/issues/349249

Previously, gnupg users had to use the option `disable-ccid` in scdaemon, to avoid scdameon conflicting with pcscd (via CCID) accessing the smart card.

However, gnupg no longer works with that option. And **rather surprisingly** does work, with the option removed, with CCID+PCSCD still enabled. For example, I can interrogate the Yubikey, via `ykman`, which I understand goes over CCID. However, `scdaemon` also seems... to work? (Maybe also using CCID)?

I'm confused. We had to add `disable-ccid` because of an scdaemon issue, as I understood it. But now it's not needed because of an update to CCID?

Can you shed any light on this? Nothing in the changelog jumped out at me. Thanks!

panchoh, 2024-10-21:
>  Where can I find the build logs of CCID 1.6.1 on NixOS? I am interested to know how the build is incorrect but the problem was not detected.

[Here](https://cache.nixos.org/log/bv28wgp3iav13rv32lzknhcd1plglzy9-ccid-1.6.1.drv)

For comparison, an actual correct build for 1.5.5 can be seen [here](https://cache.nixos.org/log/r9mrlhidrv096wzgs3mvr7qiqljlj7q3-ccid-1.5.5.drv).

(Accessed via [hydra](https://hydra.nixos.org/job/nixpkgs/trunk/ccid.x86_64-linux).)

Merci beaucoup pour ton aide, Ludovic.

colemickens, 2024-10-21:
I was unable to get the autotools build to work with 1.6.1, but that could just be my personal incompatibility with autotools, or nixpkgs autotool infra, or... something else. I notice other packagers switched to meson.

And in fact, a proposed fix moves us to meson too: https://github.com/NixOS/nixpkgs/pull/350153

At this point it's quite clearly not an issue with CCID, or even if there is still some issue, it's not the one this issue is describing. So I'm going to close this, I think the label of `invalid` is quite accurate.

>  Merci beaucoup pour ton aide, Ludovic.

I don't speak French, but I'm pretty sure I am a +1 to this. Thanks, and sorry for the noise.