LudovicRousseau/CCID #149: AK9563 procedure byte conflict

damonto, 2024-11-12:
The card works normally on other card readers, but it doesn't work on Alcor Link AK9563.

[log.txt](https://github.com/user-attachments/files/17712053/log.txt)

Here is the similar issue #142 , And I have tried to modify the code to remove the high speed baud rates, but it doesn't work normally.

[limited-baud.txt](https://github.com/user-attachments/files/17712052/limited-baud.txt)

LudovicRousseau, 2024-11-12:
No idea.
I suggest to use another reader.

It would involve some time and effort to try to make the reader work. I don't know if you have a budget for that.

damonto, 2024-11-12:
I have plenty of time to make the reader work, but I’m not sure how to do it.

LudovicRousseau, 2024-11-12:
I would start by checking if the reader works fine with Windows.
If yes, then do some USB spying to see what the Windows driver is doing.

Good luck

closed 2024-11-12 by LudovicRousseau