LudovicRousseau/CCID #150: Compatibility between CCID driver and pcsclite versions for Microchip SEC1210 URT single slot

tuananh-lab, 2024-11-14:
I want to ask which version between CCID and pcsclite are compatible with each other and support communication with IC SEC1210. I want to install pcscd service to communicate with SEC1210 URT single slot module on Android 13.
I see that Microchip's documentation says CCID version 1.4.4 and pcsclite 1.7.4, but the document is quite old and when I try it, I get an error when run pcscd s
00000000 [525255369976] debuglog.c:299:DebugLogSetLevel() debug level=debug
00000084 [525255369976] pcscdaemon.c:339:main() using new config file: /data/local/tmp/etc/reader.conf.d
00000929 [525255369976] configfile.l:293:DBGetReaderListDir() Parsing conf directory: /data/local/tmp/etc/reader.conf.d
00000065 [525255369976] configfile.l:369:DBGetReaderList() Parsing conf file: /data/local/tmp/etc/reader.conf.d/reader.smsc.conf
00000325 [525255369976] configfile.l:210:evaluatetoken() Add reader: SEC1210 Serial
00000042 [525255369976] configfile.l:330:DBGetReaderListDir() Skipping non regular file: .
00000007 [525255369976] configfile.l:330:DBGetReaderListDir() Skipping non regular file: ..
00000026 [525255369976] readerfactory.c:1075:RFInitializeReader() Attempting startup of SEC1210 Serial 00 00 using /data/local/tmp/lib/pcsc/drivers/serial/libccidsmsc.so
00000486 [525255369976] readerfactory.c:950:RFBindFunctions() Loading IFD Handler 3.0
00000071 [525255369976] ifdhandler.c:1750:init_driver() Driver version: 1.4.4
00000054 [525255369976] tokenparser.l:168:bundleParse() Could not open bundle file /data/local/tmp/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist: No such file or directory
00000039 [525255369976] ifdhandler.c:79:IFDHCreateChannelByName() lun: 0, device: /dev/ttyHS5
00001830 [525255369976] ccid_serial.c:677:OpenSerialByName() Set serial port baudrate to 115200 and correct configuration
00005987 [525255369976] ccid_serial.c:709:OpenSerialByName() Firmware: "
00005259 [525255369976] ifdhandler.c:401:IFDHGetCapabilities() tag: 0xFB3, /dev/ttyHS5 (lun: 0)
00000087 [525255369976] readerfactory.c:391:RFAddReader() Using the pcscd polling thread
00005311 [525255369976] ifdhandler.c:401:IFDHGetCapabilities() tag: 0xFAE, /dev/ttyHS5 (lun: 0)
00000067 [525255369976] ifdhandler.c:489:IFDHGetCapabilities() Reader supports 1 slot(s)
00000049 [525255369976] pcscdaemon.c:663:main() pcsc-lite 1.9.0 daemon ready.
00010632 [512524279040] ifdhandler.c:1151:IFDHPowerICC() action: PowerUp, /dev/ttyHS5 (lun: 0)
00040263 [512524279040] eventhandler.c:289:EHStatusHandlerThread() powerState: POWER_STATE_POWERED
00000193 [512524279040] Card ATR: 3B 9F 96 80 1F 87 80 31 E0 73 FE 21 1B 67 4A 4C 75 31 33 05 6E 8A 
02411801 [512524279040] ifdwrapper.c:364:IFDStatusICC() Card not transacted: 612
00000219 [512524279040] eventhandler.c:482:EHStatusHandlerThread() powerState: POWER_STATE_UNPOWERED
02003995 [512524279040] ifdwrapper.c:364:IFDStatusICC() Card not transacted: 612
00000201 [512524279040] eventhandler.c:336:EHStatusHandlerThread() Error communicating to: SEC1210 Serial 00 00
02404470 [512524279040] ifdwrapper.c:364:IFDStatusICC() Card not transacted: 612
00000200 [512524279040] eventhandler.c:336:EHStatusHandlerThread() Error communicating to: SEC1210 Serial 00 00
02404546 [512524279040] ifdwrapper.c:364:IFDStatusICC() Card not transacted: 612
00000197 [512524279040] eventhandler.c:336:EHStatusHandlerThread() Error communicating to: SEC1210 Serial 00 00
02405176 [512524279040] ifdwrapper.c:364:IFDStatusICC() Card not transacted: 612

tuananh-lab, 2024-11-14:
@LudovicRousseau If I have received the ATR, is installing the pcscd service on Android considered successful and can developers later call APIs from the PCSC standard library to develop a smart card app? Please reply me as soon as possible. Thank you!

tuananh-lab, 2024-11-14:
Why is it that when I echo the hexa command directly through the UART port, the command returns without errors, but when using the pcscd service, I get an error through the table lookup that is an An ICC is present and inactive error.

LudovicRousseau, 2024-11-14:
@tuananh-lab you have many questions and issues.
I propose to do consulting for you. Contact me at ludovic.rousseau@free.fr

tuananh-lab, 2024-11-15:
I have sent an email to ludovic.rousseau@free.fr please check me soon and respond as soon as possible.

closed 2025-02-01 by LudovicRousseau