LudovicRousseau/CCID #152: MacOSX meson updates

igarshep, 2024-11-26:
Since the switch to meson over make, I am no longer able to build the driver on OSX. I get an error saying it can't find libpcsclite. The fallback method provided does not seem to account for OSX either.

Please advise on how it's expected to work.

igarshep, 2024-11-26:
Two issues: it seems like since OSX 14 (Sonoma), SCardControl functions seem broken using the built-in driver. I understand that this has a current workaround with a defaults change, but its nice not to have to rely on Apple. 

My other complaint is the built in driver puts the Info.plist in a read only part of the OS, so being able to configure ifdDriverOptions is impossible. Being able to build the driver on OSX using the pre-meson process put the file in an editable part of the OS. This is currently my only way to get escape command support on OSX.

LudovicRousseau, 2024-11-27:
Why do you need to use escape commands?
What smart card reader do you use?

igarshep, 2024-11-27:
I currently use Identiv SPR332 V2 readers. Escape commands allow me to control the audible tone of the reader as well as control the LEDs.

I'm also incapable of installing just the Identiv driver, as that seems to break support for other readers I have.

LudovicRousseau, 2024-11-27:
You can still use the script `./bootstrap` to generate the `./configure` script and build the driver as before.