LudovicRousseau/CCID #156: Supports of HSIC Reader

septs, 2025-02-13:
<details><summary>uname -a</summary>
<p>

```console
$ uname -a
Linux septs-devel 6.13.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 08 Feb 2025 18:54:55 +0000 x86_64 GNU/Linux
```

</p>
</details> 

<details><summary>pcscd --version</summary>
<p>

```console
$ pcscd --version
pcsc-lite version 2.3.1
Copyright (C) 1999-2002 by David Corcoran <corcoran@musclecard.com>.
Copyright (C) 2001-2024 by Ludovic Rousseau <ludovic.rousseau@free.fr>.
Copyright (C) 2003-2004 by Damien Sauveron <sauveron@labri.fr>.
Report bugs to <pcsclite-muscle@lists.infradead.org>.
Enabled features: USB serial filter_names libudev polkit systemd Linux x86_64 ipcdir=/run/pcscd usbdropdir=/usr/lib/pcsc/drivers serialconfdir=/etc/reader.conf.d
MAX_READERNAME: 128, PCSCLITE_MAX_READERS_CONTEXTS: 16
```

</p>
</details>

<details><summary>lsusb -vvv</summary>
<p>

```console
$ sudo lsusb -d 1d99:0016 -vvv
Bus 003 Device 002: ID 1d99:0016 HSIC CCID-Reader
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x1d99 HSIC
  idProduct          0x0016 CCID-Reader
  bcdDevice            1.00
  iManufacturer           1 HSIC
  iProduct                2 CCID-Reader
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005d
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass        11 Chip/SmartCard
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0
      iInterface              0
[... 222 lines cut ...]
[36m00011032[0m [124817579697856] ../PCSC/src/winscard_svc.c:362:ContextThread() Authorized PC/SC client
[36m00000013[0m [124817579697856] ../PCSC/src/winscard_svc.c:365:ContextThread() Thread is started: dwClientID=10, threadContext @0x5cb2eda81780
[36m00000013[0m [124817579697856] ../PCSC/src/winscard_svc.c:383:ContextThread() Received command: CANCEL from client 10
[36m00000009[0m [124817579697856] ../PCSC/src/winscard_svc.c:871:MSGSignalClient() Signal client: 7
[36m00000002[0m [124817579697856] ../PCSC/src/winscard_svc.c:874:MSGSignalClient() SIGNAL for client 7, [31mrv=SCARD_E_CANCELLED[0m
[36m00000024[0m [124817579697856] ../PCSC/src/winscard_svc.c:666:ContextThread() CANCEL for client 10, rv=SCARD_S_SUCCESS
[36m00000089[0m [124817816925888] ../PCSC/src/winscard_svc.c:383:ContextThread() Received command: RELEASE_CONTEXT from client 7
[36m00000021[0m [124817579697856] ../PCSC/src/winscard_svc.c:376:ContextThread() Client die: 10
[36m00000017[0m [124817579697856] ../PCSC/src/winscard_svc.c:1112:MSGCleanupClient() Thread is stopping: dwClientID=10, threadContext @0x5cb2eda81780
[36m00000004[0m [124817579697856] ../PCSC/src/winscard_svc.c:1120:MSGCleanupClient() Freeing SCONTEXT @0x5cb2eda81780
[36m00000188[0m [124817816925888] ../PCSC/src/winscard.c:224:SCardReleaseContext() Releasing Context: 0x51B15438
[36m00000011[0m [124817816925888] ../PCSC/src/winscard_svc.c:516:ContextThread() RELEASE_CONTEXT for client 7, rv=SCARD_S_SUCCESS
[36m00000048[0m [124817816925888] ../PCSC/src/winscard_svc.c:376:ContextThread() Client die: 7
[36m00000013[0m [124817816925888] ../PCSC/src/winscard_svc.c:1112:MSGCleanupClient() Thread is stopping: dwClientID=7, threadContext @0x5cb2eda90660
[36m00000003[0m [124817816925888] ../PCSC/src/winscard_svc.c:1120:MSGCleanupClient() Freeing SCONTEXT @0x5cb2eda90660
```

</p>
</details> 

## Attachments

- [pcscd-log.txt](https://github.com/user-attachments/files/18781787/pcscd-log.txt)
- [lsusb-verbose.txt](https://github.com/user-attachments/files/18781789/lsusb-verbose.txt)
- [hsic-ccid-parsed.txt](https://github.com/user-attachments/files/18781811/hsic-ccid-parsed.txt)

LudovicRousseau, 2025-02-13:
This reader is not yet in my list.
I guess you edited the driver configuration file by hand to add it. Exact?

It looks like the reader works. Do you confirm?

septs, 2025-02-13:
yep, I edited driver configuration file

I confirmed is available

septs, 2025-02-14:
@LudovicRousseau can add this smart card reader to your database?

LudovicRousseau, 2025-02-14:
Yes
Please attach the file `output.bin` as documented on https://ccid.apdu.fr/#CCID_compliant

septs, 2025-02-15:
[output.bin.zip](https://github.com/user-attachments/files/18807012/output.bin.zip)

github not allowed `.bin` ext file

LudovicRousseau, 2025-02-15:
I added the reader in the "Should work but untested by me" list
https://ccid.apdu.fr/ccid/shouldwork.html#0x1D990x0016

Thanks

closed 2025-02-15 by LudovicRousseau

septs, 2025-02-18:
[image]

MCU: [nationz](https://www.nationstech.com) Z8D256U HC538004