LudovicRousseau/CCID #158: Issue with simhub pcsc reader initialization

dboyan, 2025-03-24:
The device (1677:0341) is listed as "untested, should work" as of version 1.5.5, but it doesn't initialize correctly in my local environment.

`pcscd -f` gives the following output:

```
00193388 [127711426645184] commands.c:1283:CmdGetSlotStatus Command not supported or not allowed
00000285 [127711426645184] commands.c:1283:CmdGetSlotStatus Command not supported or not allowed
00000419 [127711426645184] commands.c:1283:CmdGetSlotStatus Command not supported or not allowed
00000006 [127711426645184] ifdhandler.c:203:CreateChannelByNameOrChannel() failed
00000094 [127711426645184] ../PCSC/src/readerfactory.c:1132:RFInitializeReader() Open Port 0x200000 Failed (usb:1677/0341:libudev:0:/dev/bus/usb/001/095)
00000003 [127711426645184] ../PCSC/src/readerfactory.c:371:RFAddReader() SIMHUB pcsc reader init failed.
00000064 [127711426645184] ../PCSC/src/hotplug_libudev.c:517:HPAddDevice() Failed adding USB device: SIMHUB pcsc reader
```

This device seems to work on the default ccid driver on windows 11. I noticed that it also returns failure for the GetSlotStatus command. The windows driver somehow decides to send ICC Power Off command and then GetSlotStatus succeeds. The following is the usb trace captured by wireshark on windows, relevant usb traffic starts from packet 70.

[usbcap.zip](https://github.com/user-attachments/files/19412279/usbcap.zip)

I wonder if introducing similar logic makes sense or can be helpful, as we currently just try CmdGetSlotStatus for 3 times (and the device failed on all three).

LudovicRousseau, 2025-03-24:
Maybe a power off can fix the issue.

Try this patch:
```diff
diff --git a/src/ccid.c b/src/ccid.c
index 1948080..3bd866c 100644
--- a/src/ccid.c
+++ b/src/ccid.c
@@ -126,6 +126,10 @@ int ccid_open_hack_pre(unsigned int reader_index)
 			ccid_descriptor->dwMaxDataRate = max_speed;
 			}
 			break;
+
+		case SIMHUB:
+			CmdPowerOff(reader_index);
+			break;
 	}
 
 	/* CCID */
diff --git a/src/ccid.h b/src/ccid.h
index 11a5e88..43ff33c 100644
--- a/src/ccid.h
+++ b/src/ccid.h
@@ -290,6 +290,7 @@ typedef struct
 #define KAPELSE_KAPLIN2			0x29470105
 #define KAPELSE_KAPECV			0x29470112
 #define ACS_ACR122U				0x072f2200
+#define SIMHUB					0x16770341
 
 #define VENDOR_KAPELSE 0x2947
 #define VENDOR_GEMALTO 0x08E6
```

LudovicRousseau, 2025-03-25:
So I guess the reader does **not** support the CCID command PC_to_RDR_GetSlotStatus.
Unfortunately my driver needs this command (the windows driver is different).

I moved the reader in the "Unsupported or partly supported CCID readers" list
https://ccid.apdu.fr/ccid/unsupported.html#0x16770x0341

I suggest you to use another reader model, or contact the manufacturer.

closed 2025-03-25 by LudovicRousseau