LudovicRousseau/CCID #161: Avoiding LIBUSB_ERROR_OVERFLOW errors

bluetech, 2025-03-30:
I am trying to get the CCID driver to work with a bit finicky industrial device. Occasionally the communication with the device fails with error `LIBUSB_TRANSFER_OVERFLOW`, and then things stop working. The reason for this error is described here: https://libusb.sourceforge.io/api-1.0/libusb_packetoverflow.html

As described in the libusb doc, I was able to resolve the problem by tweaking the buffer sizes that CCID is passing to `ReadPort` (AKA `ReadUSB`) in the `commands.c` file. In principle, the needed tweak is to make the buffer sizes a multiple of the endpoint's `wMaxPacketSize`. This value can be queried from libusb. In all CCID readers I have, the value is always 64 bytes, however in the latest USB version (3.2, very unlikely to actually be used by CCID readers) it can go up to 1024 bytes if I understand correctly. So we can avoid dealing with `wMaxPacketSize` and just assume 1024.

Therefore, I'd like to propose rounding all buffer sizes used for USB bulk transfers to a multiple of 1024, or even just 64 if we want to use a lower value while still handling most devices, although 1024 doesn't seem high enough to me. I can send a PR if the idea is accepted.

LudovicRousseau, 2025-03-30:
`LIBUSB_TRANSFER_OVERFLOW` is returned whet the reader send more data than expected by the driver. It is not a problem related to `wMaxPacketSize` (unless I am warong).

If your reader is sending more data than expected by the driver then it is a bug. I would first suspect a bug in the reader.

bluetech, 2025-03-31:
Yes, this is likely a bug in the reader. The proposed change would make CCID more resilient against such hardware issues in the USB communication level.

LudovicRousseau, 2025-03-31:
OK to change the driver code.

The patch needs to be fully documented to explain why the change is needed:
- exact reader model
- logs showing the issue

fabled, 2025-05-13:
> If your reader is sending more data than expected by the driver then it is a bug. I would first suspect a bug in the reader.

Given that there is also synchronization code at:
https://github.com/LudovicRousseau/CCID/blob/a5e3ba0da1bf51419cd1027d0918869df8408c11/src/ccid_usb.c#L1143-L1155

Another valid case is that the reply is large due to being a reply to earlier sent command (e.g. which we never saw sending it due to a timeout). In this case the larger than expected reply would need to be read correctly first to be able to resynchronize. It seems the current code will just fail hard because `libusb_bulk_transfer` returns the overflow error and the resynchronization code is never reached.

LudovicRousseau, 2025-05-23:
If the driver receives a reply from a previous (n-1) command then the answer would be ignored by the wrong sequence number.
So if the answer is too big and is rejected at the libusb layer that should not be a problem.