LudovicRousseau/CCID #162: The fact that SetParameters command isn't supported by a reader should not result in falling back to T=0

rhoog-ine, 2025-04-04:
00001211 eventhandler.c:423:EHStatusHandlerThread() Card inserted into Inepro BV Spider CCID (1) 00 00
00000128 Card ATR: 3B 89 80 01 4A 43 4F 50 32 34 32 52 32 4A
05001596 ifdhandler.c:1246:IFDHPowerICC() action: PowerDown, usb:1da6/0101:libudev:0:/dev/bus/usb/001/109 (lun: 0)
01871733 ifdhandler.c:1246:IFDHPowerICC() action: Reset, usb:1da6/0101:libudev:0:/dev/bus/usb/001/109 (lun: 0)
00567138 prothandler.c:107:PHSetProtocol() Attempting PTS to T=1
00000103 ifdhandler.c:743:IFDHSetProtocolParameters() protocol T=1, usb:1da6/0101:libudev:0:/dev/bus/usb/001/109 (lun: 0)
00000606 commands.c:2351:SetParameters Command not supported or not allowed
00000052 prothandler.c:136:PHSetProtocol() Set PTS failed (614). Using T=0

The PTS isn't failed, it's handled by the reader itself. The CCID command PC_to_RDR_SetParameters isn't needed because the reader handles all parameters automatically:

(lsusb -v)

>         dwFeatures       000404BA
>           Auto configuration based on ATR
>           Auto voltage selection
>           Auto clock change
>           Auto baud rate change
>           Auto PPS made by CCID
>           Auto IFSD exchange
>           Short and extended APDU level exchange

From DWG_Smart-Card_CCID_Rev110:

>         bmICCStatus bmCommandStatus bError Description
>         0           1               0      Command not Supported

This should not mean that the PPS failed, it means that the command isn't supported and PPS is handled automatically.

rhoog-ine, 2025-04-04:
I'm sorry, I see this is already done in commit 5259925ceb74eca69b3aad7658df30d6fa938ed6

I'll download version 1.6.2 and try again.

rhoog-ine, 2025-04-04:
The change in the above commit only skips sending the SetProtocal for the _proprietary_ automatic protocol feature bit 0x40, not for the automatic negotiation 'according to the active parameters' bit 0x80. This seems to imply that those active parameters can be set using setProtocol and if setProtocol isn't supported, then it is an error. But what if those parameters cannot be set, because the reader is actually a contactless reader, which just pretends to be T=1, because that is the closest thing. The parameters for setProtocol don't really make sense for contactless, so I think it's better that the reader just says it doesn't support the command instead of lying about it and claiming it activated the parameters without actually doing anything.

So, still I would like to argue that if a reader doesn't support the setProtocol and reports that according to the CCID spec, then the pcsc daemon should not 'fall back' to T=0 gratuitously.

LudovicRousseau, 2025-04-04:
I do not have this reader "Inepro BV Spider CCID (1)" in my lists.

Please provide a full debug trace as documented in https://ccid.apdu.fr/#support with the latest CCID version.

rhoog-ine, 2025-04-04:
Right, that is the reader I'm working on. The CCID mode is quite new. Once I'm in a stable state, then I will provide the debug trace you ask about.

The argument is about whether, generally, it is appropriate for the pcsc daemon to fall back into T=0 if a CCID reader that claims to handle all PPS automatically but doesn't support the setProtocol command. Maybe I will eventually be forced to lie about setProtocol support, but I would prefer not to.

The windows pcsc re-tries the setProtocol command if the reader reports no support (which is weird if you think about it), so the second time the same command I don't support arrives, I will just say all is fine, while in fact I didn't do anything with the provided parameters.

LudovicRousseau, 2025-04-04:
The protocol selection is done by pcsc-lite, not the CCID driver.
Have a look at https://github.com/LudovicRousseau/PCSC/blob/master/src/prothandler.c#L55

Maybe there is a bug in my code.

LudovicRousseau, 2025-04-05:
What is the problem is T=0 is used instead of T=1?

LudovicRousseau, 2025-04-21:
Hello @rhoog-ine I have no answer to my question.

Can you explain the problem when/if T=0 is used?

rhoog-ine, 2025-04-22:
In principle theoretically not much, since the reader uses APDU level and APDU is APDU no matter wat low level protocol is used. However in practice the application might have incorrect expectations, especially when extended APDUs are concerned. It might wrap extended C-APDU in ENVELOPE commands and expect R-APDUs no longer than 256 bytes per segment, expecting 61xx status if there is more.

LudovicRousseau, 2025-04-22:
Do you think there is a problem to be fixed here?
If yes can you propose a patch?

closed 2025-06-16 by LudovicRousseau