LudovicRousseau/CCID #165: Question on Alcor  AU9540 card reader

choman, 2025-04-21:
I have a laptop with an Alcor AU9540 card reader, this worked with chrome on ubuntu 24.04 and 24.10. I recenrly upgraded to 25.04
which brings me to libccid 1.6.1. After reboot the pcsc_scan works as expected. but when i fire up 
    - firefox (deb), firefox sees the device  as disabled. 
    - chrome seems to mostly but the device into a stuck state

I've tried two differenent SMC readers and they seem to work fine, but on a laptop they get in the way. 

I am also planning to roll back to the snap firefox and try out the "connector" for pcscd. 

Anyways, enough for now 
Cheers

LudovicRousseau, 2025-04-22:
If pcsc_scan works fine then it indicates the pcsc-lite + driver works as expected.

You tried the external readers with Firefox? Is it working fine?

LudovicRousseau, 2025-04-22:
OK.
Generate logs as described in https://ccid.apdu.fr/#support

LudovicRousseau, 2025-06-16:
No news since 2 months.
Closing.

closed 2025-06-16 by LudovicRousseau