LudovicRousseau/CCID #166: Synchronous libusb API and occasionally lost packets

0xfeed-code, 2025-04-28:
We've been having an issue with our custom multi-slot card reader and PCSCD + PCSC-Lite. Occasionally (roughly once a week on 5 devices running), the application using PCSC-Lite would detect a smart card disconnection, because the returned state is "UNAVAILABLE" (see also !163 for more background). Waiting a few seconds to wait for it fixing itself does not help (at least with the PCSCD default configuration).

As it turns out, there is a `PC_to_RDR_GetSlotStatus` request to the reader, that PCSCD sends but does not receive an answer for it. However, when using `usbmon` and tcpdump, we can clearly see the response sent over the USB bus from the card reader. The log from the card reader also indicates that the packet (not sure if packet is the right term here) is sent back. What seems to happen at the same time when we loose the packet, is a [timeout from `libusb_bulk_transfer()` in PCSCD's `Multi_ReadProc()`](https://github.com/LudovicRousseau/CCID/blob/master/src/ccid_usb.c#L2149). It looks a bit as if in between the return and restart of `libusb_bulk_transfer()`, there is a short moment when USB packets can get lost, which would explain why it happens so rarely.

I'm not very familiar with the libusb API, but I found a message on the libusb mailing list, suggesting that the synchronous API cannot be used for continuous transfers: https://libusb-devel.narkive.com/1AunCCLY/libusb-bulk-read-losing-data-on-read-buffer-swap#post3.

I have the logs and usb capture file, but need to check them first to avoid leaking unwanted data.

LudovicRousseau, 2025-04-28:
Why haven't you followed the template to create the issue?

LudovicRousseau, 2025-05-07:
I think I see the (potential) problem.

Only 6.65% of readers are multi-slots. https://ccid.apdu.fr/select_readers/?bMaxSlotIndex%E2%89%A51
And I do not have access to many of them.
So a rework + testing of the code will be difficult.
But your company could help.

0xfeed-code, 2025-05-07:
@LudovicRousseau I changed the post above to hopefully suit the template a bit more. We're actually thinking about implementing a fix, but we haven't decided how to handle the problem. Before we start implementing, however, it would be great to have an assessment if our assumptions and analysis above are entirely correct. Just to avoid implementing a fix that does not fix.

LudovicRousseau, 2025-05-08:
First, upgrade the CCID driver to the latest version: 1.6.2, and try again.

LudovicRousseau, 2025-05-14:
I am not sure this issue is fixed.
@0xfeed-code please test the current code and tell me if the problem is still present.

PR #168 (2025-05-14): multi slots: fix potential race in interrupt handling
Initialize completed before calling libusb_submit_transfer. Otherwise it is possible that the callback fires in another thread first, and the completed=1 assignment gets lost.

Also use volatile int for the completed flag to hint that it can change from another thread.

Potentially fixes #166
closed 2025-05-14 by LudovicRousseau